hybridgroup/gobot #661: BB-8 examples won't connect

mrubelmann, 2019-03-29:
I'm trying to run both of the BB-8 example programs on macOS, but it won't connect to the device.  It gets as far as trying to start the connection, but then never makes it into the work() function.  I added `println("In work")` to the top of work(), but as you can see, it never gets called:

```
bash-3.2$ GODEBUG=cgocheck=0 go run examples/bb8.go BB-FCOB
2019/03/28 23:37:55 Initializing connections...
2019/03/28 23:37:55 Initializing connection BLEClient-76ABEF2F2928F188 ...
2019/03/28 23:37:55 Initializing devices...
2019/03/28 23:37:55 Initializing device BB8-6C8E7A10A90D57A9 ...
2019/03/28 23:37:55 Robot bbBot initialized.
2019/03/28 23:37:55 Starting Robot bbBot ...
2019/03/28 23:37:55 Starting connections...
2019/03/28 23:37:55 Starting connection BLEClient-76ABEF2F2928F188...
```

Is it possible that the firmware has changed and the gobot driver no longer works, or am I just doing something wrong?

mrubelmann, 2019-04-16:
It turns out this is a Mac issue.  I couldn't get it to work on Mojave (10.14.4), but it's working on Ubuntu.

deadprogram, 2019-05-31:
It would be great to get some people to collaborate on macOS BLE stack updates for Go. Anyone who might want to help out and knows CGo please make yourself known! :)

deadprogram, 2019-10-25:
The https://github.com/raff/goble repo has been updated for macOS Catalina.

Also this fork of the original `gatt` package seems active https://github.com/bettercap/gatt

Bit of history, Gobot originally used gatt... seems funny to imagine going back.

Just wondering out loud if anyone would be able to look at replacing `go-ble` in Gobot with this fork of `gatt`?

gen2thomas, 2024-11-01:
Hi @mrubelmann , @deadprogram can we close this issue, maybe because we has updated our BLE drivers or Mojave is out of interest?

Thanks Thomas

deadprogram, 2024-11-01:
Closing as no longer needed.

closed 2024-11-01 by deadprogram