hybridgroup/gobot #663: SSD1306 Screen garbled

TinHead, 2019-04-11:
Hello, 

I know you can't possibly support all variants of those displays but here we go.

I have one of these: 
https://www.aliexpress.com/item/0-96-Inch-SPI-OLED-Display-Module-White-blue-color-128X64-OLED-7Pin-Yellow-blue/32850341305.html?spm=2114.search0104.3.37.5dc71e51kDr2dR&ws_ab_test=searchweb0_0%2Csearchweb201602_7_10065_10068_10890_10546_319_10059_10884_10548_317_10887_10696_321_322_453_10084_10083_454_10103_10618_10304_10307_10820_537_536%2Csearchweb201603_80%2CppcSwitch_0&algo_pvid=7d44f833-92d0-44a1-a135-d2a4485f9e65&algo_expid=7d44f833-92d0-44a1-a135-d2a4485f9e65-5

The blue one and yes *spi* setup by default. I have modded it to i2c by moving r3 resistor to r1 and bridging r8. 

I have tested it with adafruit https://github.com/adafruit/Adafruit_SSD1306 and it worked flawlessly after setting the i2c address to 0x3c, as opposed to the default 0x3d on the original ones.

Then I have tried it with gobot using the same arduino nano running firmata and the ssd1306 i2c driver in the master branch, and after fiddling with toggling the RES pin to reset the display I had it turn on and display .... completely random pixels. And this is where I'm stuck.

Questions: 
1. In the datasheet and the adafruit code the 0x00 byte is used to signal a command and 0x40 to signal data however in gobot we send 0x80 for command ... Why?
2. Transfer speed seems a lot slower than adafruit looking at the scope ... 100kHz vs 400kHz is i2c bus setup to 100kHz in Gobot? Could not find anywhere.
3. Anyone else seen these issues? 

Thank you,
Razvan

deadprogram, 2019-05-31:
You might want to try the latest release of Gobot as there are several commits from @ulisesflynn specific to these displays.

The I2C bus speed is usually determined on the host system, and depends on what it can support. The devices are usually more tolerant of different speeds/volatages than the hosts are capable of varying.

Anyhow, hope that helps. I use a couple of variations of these displays with no problems, but YMMV.

TinHead, 2019-05-31:
Hi, 
OK I'll try the updated version, also next week I'll be able to try some factory I2C displays which I have ordered.

Thanks for your reply and keep up the good work :)

TinHead, 2019-07-14:
Closing as i don't really had much time to play.

closed 2019-07-14 by TinHead