hybridgroup/gobot #665: Is gobot project dead?

tommyblue, 2019-05-02:
Hi all, looking at commits, releases and Google goup, it seems the project is quite dead.
Long time requests to add new platforms seem to be completely ignored.
Is someone working on this project? Is it dead or alive?
Thanks!

deadprogram, 2019-05-02:
Hi @tommyblue you might want to look at the `dev` branch, that is where all the activity is.

https://github.com/hybridgroup/gobot/tree/dev

Hope that helps!

closed 2019-05-03 by deadprogram

rosscdh, 2021-04-30:
Once again is the project dead? dev branch last commit was 3+ years ago?

deadprogram, 2021-04-30:
The most recent commit to `dev` branch was https://github.com/hybridgroup/gobot/commit/27c76b4585807e829965ec6a4551cadc28f430fe which was 16 days ago.

rosscdh, 2021-04-30:
Look closer he hears ;) thanks

tommyblue, 2021-04-30:
indeed it's still a good question 😄 
as a end user, since my initial question, all the platforms I had the chance to play with, aren't supported by gobot (judging from the project home page): sphero mini, esp32, parrot mambo and (correct me if I'm wrong) it doesn't seem to have gained support to many platforms lately

deadprogram, 2021-04-30:
Sphero Mini support is requiring quite a few changes, as the API changed.

ESP32 you can use firmata right now just as you can with the ESP8266.

Parrot Mambo, I do not have one of those myself. Several people have been working on automating the generation of the wrappers based on the Parrot XML files, but that work has yet to land.

Any other specific hardware support you are looking for @tommyblue?

rosscdh, 2021-04-30:
rpi4 seems to have issues with gpio too

tommyblue, 2021-04-30:
> Any other specific hardware support you are looking for @tommyblue?

nope, I was looking for Sphero mini but I got the answer.
As per ESP32, good to know it's supported, maybe it's worth adding the entry in the platform list in the documentation?