hybridgroup/gobot #669: Gobot build release version

adrianowead, 2019-05-26:
How can I build the "release" version?

My code run's sucessfully on arduino, but only works on serial connection.

I didn't found on docs how to compile and install the "release" version of my program, to run standalone without any connection.

deadprogram, 2019-05-31:
TO run Go code directly on a microcontroller, you would need to use TinyGo (https://tinygo.org/) or something like that. Gobot only executes on host computers, you might think of it as middleware or application layer in your architecture.

Hope that helps!

closed 2019-05-31 by deadprogram

adrianowead, 2019-05-31:
> TO run Go code directly on a microcontroller, you would need to use TinyGo (https://tinygo.org/) or something like that. Gobot only executes on host computers, you might think of it as middleware or application layer in your architecture.
> 
> Hope that helps!

Thank you very much!