hybridgroup/gobot #684: github.com/gobuffalo/uuid syntax error on Raspbian

paul-wolf, 2019-07-21:
Trying to run example on Raspberry PI 3 from <https://gobot.io/documentation/getting-started/> that makes an LED light blink and getting this error:

```
# github.com/gobuffalo/uuid
../../go/src/github.com/gobuffalo/uuid/generator.go:44: syntax error: unexpected = in type declaration
```

Am I missing something? 

System info:
```
pi@raspberrypi:~/prj/gobot $ cat /etc/debian_version
9.4
pi@raspberrypi:~/prj/gobot $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
pi@raspberrypi:~/prj/gobot $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a22082
Serial          : 00000000d8172f8c
```

deadprogram, 2019-08-12:
Perhaps is a breaking change in the `uuid` package. You might want to install the v2.0.5 version as is specified here https://github.com/hybridgroup/gobot/blob/master/Gopkg.lock#L64

paul-wolf, 2019-08-13:
Thanks, @deadprogram. I appreciate it.

closed 2019-08-13 by paul-wolf

redmercury, 2019-09-08:
Sounds like it might be a go version issue?

https://github.com/google/go-jsonnet/issues/172