hybridgroup/gobot #685: GPIO already exported

akhudek, 2019-07-29:
On the Beaglebone Green it turns out that GPIO pins start already exported. This causes Gobot to fail on the first run because it can't export the already exported pin. However, once Gobot shuts down, it unexports the pins and subsequently works on being restarted.

This is easy enough to work around by manually unexporting pins on device boot, but it would be nice if Gobot either more gracefully handled this situation or at least provided a way for the user to handle it.

akhudek, 2019-08-04:
Actually, this is way more complicated. Turns out the beaglebone exports all the pins by default, and unexport doesn't work, it just errors out. But, if you try to export an already exported pin, *it unexports* it. Obviously unwanted behaviour. 

I have a patch for gobot that checks if the pin is already exported, but it's also more complicated than I thought due to some fixed needed to the gobot testing system. Not sure if it's worth doing a PR for this given it should probably be fixed in the beaglebone firmware.

gen2thomas, 2024-11-01:
Hi @akhudek , do you know, if this is fixed now in BB green?

Thanks Thomas

akhudek, 2024-11-01:
No clue, it's been years since I've used a beaglebone at this point. Usually I just get pi's now.

gen2thomas, 2024-11-01:
Thanks for the fast response, so I'm going to close this issue, because out of interest.

closed 2024-11-01 by gen2thomas