hybridgroup/gobot #691: How do I use gobot to control servo on raspberry pi?

erkexzcx, 2019-09-15:
Hi. Spent several hours trying to find any tutorial or documentation on how to control servo on raspberry pi using gobot.

No documentation there: https://gobot.io/documentation/drivers/servo/

Nothing is working from here either: https://github.com/hybridgroup/gobot/issues/166 (tried code from there as well).

Currently trying this code, but nothing is working. My C code works just fine on raspberry pi to control servo, but can't get it to work on gobot. This is the code I am trying to use:
```go
package main

import (
        "log"
        "time"

        "gobot.io/x/gobot"
        "gobot.io/x/gobot/drivers/gpio"
        "gobot.io/x/gobot/platforms/raspi"
)

func main() {
        r := raspi.NewAdaptor()
        servo := gpio.NewServoDriver(r, "18")
        work := func() {
                gobot.Every(2*time.Second, func() {
                        i := uint8(gobot.Rand(180))
                        log.Printf("Moving servo to %d", i)
                        servo.Move(i)
                })
        }

        robot := gobot.NewRobot("blinkBot",
                []gobot.Connection{r},
                []gobot.Device{servo},
                work,
        )

        robot.Start()
}
```
Output:
```
2019/09/15 17:47:31 Initializing connections...
2019/09/15 17:47:31 Initializing connection RaspberryPi-FDF7316 ...
2019/09/15 17:47:31 Initializing devices...
2019/09/15 17:47:31 Initializing device Servo-AD871A0 ...
2019/09/15 17:47:31 Robot blinkBot initialized.
2019/09/15 17:47:31 Starting Robot blinkBot ...
2019/09/15 17:47:31 Starting connections...
2019/09/15 17:47:31 Starting connection RaspberryPi-FDF7316...
2019/09/15 17:47:31 Starting devices...
2019/09/15 17:47:31 Starting device Servo-AD871A0 on pin 18...
2019/09/15 17:47:31 Starting work...
2019/09/15 17:47:33 Moving servo to 100
2019/09/15 17:47:35 Moving servo to 47
2019/09/15 17:47:37 Moving servo to 3
2019/09/15 17:47:39 Moving servo to 80
```
But servo is not moving. It is connected it to `GPIO18`.

Any help is appreciated.

erkexzcx, 2019-09-16:
What I found out when I tested go program using LED that I need to use is index number of pin, not GPIO pin number.

For example:
- `GPIO18` - incorrect
- `18` - incorrect
- `12` - **OK**

But that still doesn't change the fact that I can't get servo to work...

deadprogram, 2019-09-16:
Please see our docs page https://gobot.io/documentation/platforms/raspi/ for the section labeled "Enabling PWM output on GPIO pins."

Hope that helps!

erkexzcx, 2019-09-16:
So these are the instructions:
1. Set-up [pi-blaster](https://github.com/sarfata/pi-blaster) on Raspberry pi.
2. Connect servo to Raspberry Pi, to pin 12 (`GPIO18`). Note that `gobot` uses pin index numbers.
3. Create file `servo.go` on your raspberry pi:
```go
package main

import (
        "log"
        "time"

        "gobot.io/x/gobot"
        "gobot.io/x/gobot/drivers/gpio"
        "gobot.io/x/gobot/platforms/raspi"
)

func main() {
        adaptor := raspi.NewAdaptor()
        servo := gpio.NewServoDriver(adaptor, "12")

        work := func() {
                gobot.Every(1*time.Second, func() {
                        i := uint8(gobot.Rand(180))
                        log.Printf("Moving servo to %d", i)
                        servo.Move(i)
                })
        }

        robot := gobot.NewRobot("Servo robot",
                []gobot.Connection{adaptor},
                []gobot.Device{servo},
                work,
        )

        robot.Start()

}
```
4. Compile it `go build servo.go`.
5. Run it `./servo.go`. Try it as root if it doesn't work (I have this issue) with command `sudo ./servo.go`.

Servo works, but somewhat not as expected (randomly doesn't work), but this is different issue anyway.

closed 2019-09-16 by erkexzcx