hybridgroup/gobot #692: Raspi servo does not work as expected

erkexzcx, 2019-09-16:
**Description**

So servo should do a 180 degrees spin using this code:
```go
package main

import (
	"time"

	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/raspi"
)

func main() {
	adaptor := raspi.NewAdaptor()
	adaptor.Connect()

	servo := gpio.NewServoDriver(adaptor, "12")
	servo.Start()

	for i := uint8(0); i <= 180; i++ {
		servo.Move(i)
		time.Sleep(100 * time.Millisecond)
	}
}
```
To be exact, this below code should do a 180 degree servo spin:
```
for i := uint8(0); i <= 180; i++ {
	servo.Move(i)
	time.Sleep(100 * time.Millisecond)
}
```
So according to documentation, `raspi.NewAdaptor()` provides function `Move(uint8)` that expects numbers from 0 to 180 (represents a servo degree).

Unfortunately, my servo from AliExpress (says it's _Tower Pro Micro Servo 9G SG90_) does a full rotation on a range between 9 to 45, so range from 0 to 180 is useless and does not work correctly.

**Expected behavior**

Servo fully rotates to one side when I set `Move(0)` and should do 180degree spin when I set `Move(180)`.

**Actual behavior**

Servo doesn't do anything when I set `Move(0)` and doesn't do anything when I set `Move(180)`.

Instead, It fully rotates to one side when I set `Move(9)` and fully rotates to another side when I set `Move(45)`.

**Additional notes**

Using _Raspberry Pi Zero W_ with _Arch Linux_.

`pi-blaster` installed, configured and **working as expected** (if used from bash). https://github.com/sarfata/pi-blaster

P.S. _A bit_ related, but figured out everything on that one. https://github.com/hybridgroup/gobot/issues/691

erkexzcx, 2019-09-16:
Below code works as expected:
```go
package main

import (
	"time"

	"gobot.io/x/gobot/platforms/raspi"
)

func main() {
	adaptor := raspi.NewAdaptor()
	adaptor.Connect()

	for i := uint8(0); i <= 180; i++ {
		adaptor.ServoWrite("12", i)
		time.Sleep(40 * time.Millisecond)
	}
}
```

Why is everything so complicated? :(

erkexzcx, 2019-09-16:
This is likelly my own issue. `Tower Pro SG90` has pulse width of `500-2400 µs`.

Is it possible to change it?

pancakestack-io, 2022-03-19:
hi @erkexzcx 

i know it has been super long since you posted this - but this helped me solve my issues. i can also only rotate 45 which ends up being 180 degrees. i am not sure if it is that the smaller servos only move in 4 degree increments or what.  if you managed to change it to move 180 degree angle let me know or if you kept it at every 1 = 4 degrees

gen2thomas, 2022-09-20:
It looks like the "servo_driver.go" needs additional parameters, e.g. NewServoDriver(a, pin, minValue, maxValue) or, much better, we should add the options functionality like i2c drivers have (e.g. "WithLimits(min, max)").

I think this would be a nice enhancement.

gen2thomas, 2023-11-10:
I'm going to close this issue as a duplicate, because it is now referenced by an issue which collect all the various problems with raspi and gpio.servo.

PR #1046 (2023-12-05): adaptors(pwm): introduce scale option for servo
## Solved issues and/or description of the change

fix #692 and a step for #1032 , this means:
* option for change range of duty-cycle (pin specific)
* option for change range of angle (pin specific)
* option to change default period (pin specific)

additionally:
* change architecture for existing PWM options to the "applier" (see aio and gpio implementations), away from func(option) architecture.
* improve PWM related tests for all platforms

## Manual test

- OS and Version (Win/Mac/Linux): tinkerboard
- Adaptor(s) and/or driver(s): 
  - [x] mo [... cut]
closed 2023-11-10 by gen2thomas