hybridgroup/gobot #698: Example rotary sensor code on grovepi with raspi doesn't work

didrocks, 2019-09-27:
With latest gobot v1.13.0.

Trying to run the example rotary sensor code doesn't seem to work: https://github.com/hybridgroup/gobot/blob/master/examples/raspi_grove_pi_rotary.go.
My rotary sensors is connected to A1 as well, and no callback for the "On" event is called when rotating it.

Note that the rotary sensor is functional, a minimal program using `grovepi.go` directly works:
```golang
package main

import (
	"fmt"
	"log"
	"time"

	"github.com/didrocks/foo/grovepi"
)

func main() {
	var g grovepi.GrovePi
	g = *grovepi.InitGrovePi(0x04)
	err := g.PinMode(grovepi.A1, "input")
	if err != nil {
		fmt.Println(err)
	}

	t := time.NewTicker(time.Second)
	defer t.Stop()
	for {
		select {
		case <-t.C:
			i, err := g.AnalogRead(grovepi.A1)
			if err != nil {
				log.Printf("couldn't read value from rotary sensor: %v\n", err)
			}
			fmt.Println(i)
		}
	}
}
```

However, you can't mix this code of code with gobot, as after initializing gobot, the values returned by an `g.AnalogRead()` call always return the max value (I think gobot initialization change some gpio operation state?).

erkexzcx, 2019-12-08:
https://github.com/hybridgroup/gobot/issues/692

It's broken - this framework is useless for me. Instead I am using below code for my automation:

```
package main

import (
        "fmt"
        "log"
        "net/http"
        "time"

        piblaster "github.com/ddrager/go-pi-blaster"
)

const pin = 18 // GPIO18
var blaster = piblaster.Blaster{}

func main() {

        // Initialize blaster
        blaster.Start([]int64{pin})

        // Do a servo routine
        for i := 0; i < 7; i++ {
                setAngle(142)
                time.Sleep(600 * time.Millisecond)
                setAngle(129)
                time.Sleep(120 * time.Millisecond)
        }
        setAngle(48)

}

// Move servo to requested angle
func setAngle(angle uint8) {
        blaster.Apply(pin, angleToServo(angle))
}

// helper function - converts angle to servo value (aka 'map' function)
func angleToServo(val uint8) float64 {
        return (float64(val)-0)*(0.25-0.05)/(180-0) + 0.05
}

```

gen2thomas, 2024-11-01:
Hi @didrocks , the driver was reworked a bit in the last years. Do you be able to re test it?

Thanks Thomas

gen2thomas, 2025-02-15:
I'm going to close this issue due to no response and maybe it is working now after heavily rework.

closed 2025-02-15 by gen2thomas