hybridgroup/gobot #703: Can't move servo using raspberry pi adapter and servo driver

ezerw, 2019-10-26:
Here is my code 

```
package main

import (
	"log"
	"time"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/raspi"
)

func main() {
	r := raspi.NewAdaptor()
	servo := gpio.NewServoDriver(r, "11")
	work := func() {
		gobot.Every(10*time.Second, func() {
			i := uint8(gobot.Rand(180))
			log.Printf("Moving servo to %d", i)
			servo.Move(i)
		})
	}
	robot := gobot.NewRobot(
		"servoBot",
		[]gobot.Connection{r},
		[]gobot.Device{servo},
		work,
	)

	robot.Start()
}

```

I've also have installed and running `pi-blaster` but still not able to move the servo. The program runs and I can see the logs but it seems that the signal is not being sent to the servo.

I tried using different pins and also following the indications on pi-blaster repo (`You must use the GPIO number (BCM xx in the diagram below).` but nothing works.

Any ideas how to debug or what's going on?

here the output of `pi-blaster`:

```
pi@raspberrypi:~/pi-blaster $ sudo ./pi-blaster
MBox Board Revision: 0xa020d3
DMA Channels Info: 0x7f35, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                32
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x3f007000
Initialised, Daemonized, Reading /dev/pi-blaster.
```

erkexzcx, 2019-12-08:
https://github.com/hybridgroup/gobot/issues/692

It's broken - this framework is useless for me. Instead I am using below code for my automation:

```
package main

import (
        "fmt"
        "log"
        "net/http"
        "time"

        piblaster "github.com/ddrager/go-pi-blaster"
)

const pin = 18 // GPIO18
var blaster = piblaster.Blaster{}

func main() {

        // Initialize blaster
        blaster.Start([]int64{pin})

        // Do a servo routine
        for i := 0; i < 7; i++ {
                setAngle(142)
                time.Sleep(600 * time.Millisecond)
                setAngle(129)
                time.Sleep(120 * time.Millisecond)
        }
        setAngle(48)

}

// Move servo to requested angle
func setAngle(angle uint8) {
        blaster.Apply(pin, angleToServo(angle))
}

// helper function - converts angle to servo value (aka 'map' function)
func angleToServo(val uint8) float64 {
        return (float64(val)-0)*(0.25-0.05)/(180-0) + 0.05
}

```

gen2thomas, 2023-11-10:
I'm going to close this issue as a duplicate, because it is now referenced by an issue which collect all the various problems with raspi and gpio.servo.

closed 2023-11-10 by gen2thomas

gen2thomas, 2023-12-09:
> PWM frequency:                 100 Hz

* pi-blaster needs to be set to 50Hz for most servos, otherwise the servo is not working at all or not working stable
* the current implementation in gobot maybe use the wrong range for the servo, this is fixed with #1046