hybridgroup/gobot #711: You are just a STOP-BOT

StefanL38, 2019-12-14:
the https://gobot.io -homepage offers no contact possability at all. So I chosed this way to get in contact with the gobot-creators. 
Admin of this part of github if you know a better way to transport this feedback to the gobot-makers please do so and delete this issue as it is no classcial github-issue at all. I'm aware of that.

gobot-makers: Do you really think that highly sophisticated short explanations like 

"go get -d -u gobot.io/x/gobot/..."  are an EASY way for totally newbees to learn about coding??

NO WAY!! . And the fact that you don't offer any feedback-possability on your website is showing that you are just nerds that are too lazy to step down from the github-heaven down to earth for really introductional material.

You should rename some parts on your homepage to say the truth:
We offer a platform for a lot of devices on a professional level. People who would like to work with this should have a good knowledge about linux, installing packages, etc. 
it is NOT convenient for beginners.

ohiaia, 2020-01-03:
OK, it seems to me that you could have open an issue about the documentation.

And I would have agreed with you. A better example, some bits of code here and there (I had to search a lot, google has been my friend, right, but not a very close one! and I succeeded just because I was really curious and stubborn) in the documentation would be great! 

But: there's no need to be so aggressive mate!

deadprogram, 2020-01-11:
Closing due to being too mean. Sorry if it was not all you hoped for.

closed 2020-01-11 by deadprogram