hybridgroup/gobot #712: Run function of StepperDriver has no sleep

tdaira, 2019-12-22:
Run function of StepperDriver does not working properly because there is no Sleep function.
This function calls step function very first cycle and should have a sleep like Move function.

Current Move function which has a sleep: https://github.com/hybridgroup/gobot/blob/master/drivers/gpio/stepper_driver.go#L214

Current loop of Run function without Sleep:
https://github.com/hybridgroup/gobot/blob/master/drivers/gpio/stepper_driver.go#L127

deadprogram, 2020-01-11:
Hi @tdaira I think the idea there is that `Run()` would just keep turning the stepper continuously. 

I thought that the pause in `Move()` was to ensure that the stepper only moves by the requested number of steps.

Probably adding any very short delay into `Run()` would address your issue. If you were to try that and it works perhaps you would be so kind as to submit a PR with the change.

gen2thomas, 2022-08-25:
@deadprogram the fix is merged already, can we close the Issue?

closed 2022-09-02 by deadprogram