hybridgroup/gobot #717: Platform request: ESP32

theoretick, 2019-12-31:
I'd love to see support for ESP32 at some point. Based on https://github.com/hybridgroup/gobot/issues/301 it sounds like it might be straightforward once the upstream support for ESP32 is handled within firmata https://github.com/firmata/arduino/issues/315, but figured I'd open this issue to track this request and see if I'm missing  anything else.

Thank you for gobot ❤️

deadprogram, 2020-01-11:
Hi @theoretick I think your assessment sound correct regarding Firmata on the ESP32.

Probably everything that applies to using the ESP8266 would be the same for the ESP32 if using WiFiFirmata.

Please let us all know if you get to try this out, and how it goes.

nilsmagnus, 2021-02-04:
@theoretick have you had a chance to try this out yet?

theoretick, 2021-02-04:
> @theoretick have you had a chance to try this out yet?

@nilsmagnus no, I haven't. I was simply waiting on https://github.com/firmata/arduino/issues/315 to close, but it does look like there's a test branch available from `soundanalogous` if you're interested in trying that out. I haven't really dug into setting anything up

gen2thomas, 2023-07-07:
Hi @theoretick , @nilsmagnus and @deadprogram the [issue](https://github.com/firmata/arduino/issues/315) is now closed with the comment "ESP32 is supported for ConfigurableFirmata." Should we close our issue now or is there anything to do in gobot?

Thanks Thomas

tlindsay, 2023-09-12:
@gen2thomas According to [this comment](https://github.com/hybridgroup/gobot/issues/665#issuecomment-829943748), ESP32 is already supported? Not sure if it's accurate or not.

gen2thomas, 2023-09-13:
Hi @tlindsay , it seems nobody has tested this yet. Could you do this? I'm not experienced with firmata.

gen2thomas, 2025-02-15:
I'm going to close this issue, because is should work in general with firmata.

closed 2025-02-15 by gen2thomas