hybridgroup/gobot #733: How to read the value of MCP3002 via SPI on Raspberry Pi

ngktt, 2020-03-09:
I tried to read the temperature sensor value using MCP3002 via SPI on Raspberry Pi but it didn't work.

I ran the following code but got the error:
```golang
package main

import (
	"log"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/platforms/raspi"
	"gobot.io/x/gobot/drivers/spi"
)

func main() {
	r := raspi.NewAdaptor()
	d := spi.NewMCP3002Driver(r)

	gobot.NewRobot("robot",
		[]gobot.Connection{r},
		[]gobot.Device{d},
		func() {
			log.Println(d.AnalogRead("19"))
		},
	).Start()
}
```
The hardware layout:
 [image]

gen2thomas, 2022-09-01:
If this is still relevant, please could you provide the output (the error). Otherwise we can't do or recomment anything without having the exact same hardware.

gen2thomas, 2023-01-28:
I will going to close this issue, because a huge rework was done for SPI. Possibly the original failure is gone. Otherwise, please test with dev branch first and reopen or file a new one.

closed 2023-01-28 by gen2thomas