hybridgroup/gobot #741: go get -d -u gobot.io/x/gobot/... on windows failing

Sumanth-R-Y, 2020-03-28:
go version:
go version go1.14.1 windows/amd64

>go get -d -u gobot.io/x/gobot/...
**# cd xxxxxxxxx\go\src\go.bug.st\serial.v1; git pull --ff-only
fatal: not a git repository (or any of the parent directories): .git
package go.bug.st/serial.v1: exit status 128**

Sumanth-R-Y, 2020-04-05:
Tried today and it is working.

urban-wombat, 2020-10-29:
I'm seeing something similar:

$ go get -d -u gobot.io/x/gobot/...
# cd xxxxxxxxxx\golang\src\gobot.io\x\gobot; git pull --ff-only
fatal: not a git repository (or any parent up to mount point /cygdrive)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
package gobot.io/x/gobot/...: exit status 128

pinnaclesystemsgroup, 2023-03-13:
Could SOMEONE PLEASE UPDATE THE SITE'S WEBPAGE AND MAKE IT THE INSTALL COMMAND CONSISTENT WITH THE GOLANG STANDARD THAT DEPRECATED THE "go get -d -u gobot.io/x/gobot/.." install command!!!!

gen2thomas, 2023-03-13:
Hi @pinnaclesystemsgroup , thanks for pointing this out. PR's are greatly appreciated. So at least for the README.md this can be fixed by anyone.

gen2thomas, 2023-06-04:
The CLI was removed and the documentation was updated. Currently there is a [known issue](https://github.com/hybridgroup/gobot/issues/920) with v2+ modules.

closed 2023-07-07 by gen2thomas