hybridgroup/gobot #744: Can't install Gobot CLI

DrunkenPoney, 2020-04-03:
I get an error when I try to install the gobot CLI.
[image]

What am I doing wrong?

OS: Ubuntu 19.10
Go version: 1.14

pseidemann, 2020-04-19:
you have `/gobot` twice in the path

DrunkenPoney, 2020-04-22:
In fact, I did exactly what the [documentation](https://gobot.io/documentation/guides/cli/) says...
[image]

Anyway, it still doesn't work even with a single `/gobot` in the path.
[image]

pseidemann, 2020-04-22:
seems like a mistake. did you try https://github.com/hybridgroup/gobot/#getting-started?

DrunkenPoney, 2020-05-02:
I don't have any problem with the library. I just can't install the CLI.

pseidemann, 2020-05-03:
this should install the cli to the default go binaries directory as "gobot":
```go build -o "$(go env GOPATH)/bin/gobot" gobot.io/x/gobot/cli```
(derived from https://github.com/hybridgroup/gobot/blob/master/cli/README.md)

DrunkenPoney, 2020-05-05:
It worked! Thanks! However, the documentation still need to be fixed.

ramoncodes-collab, 2020-05-10:
It seems the documentation is not updated as @DrunkenPoney  mentions. Would be good if its updated

deadprogram, 2020-05-10:
We'd gladly accept a PR that corrects the docs! :smile_cat:

gen2thomas, 2023-06-04:
The CLI was removed and the documentation was updated. Currently there is a [known issue](https://github.com/hybridgroup/gobot/issues/920) with v2+ modules.

closed 2023-07-07 by gen2thomas