hybridgroup/gobot #753: Servo doesn't work on raspberry pi

nichoriverae, 2020-06-07:
I am having trouble getting a servo to move on a raspberry pi using the go1.13. I have never been able to get this to work. Is there something I'm missing? 
```
package main
import (
        "time"
	"fmt"

        "gobot.io/x/gobot"
        "gobot.io/x/gobot/drivers/gpio"
        "gobot.io/x/gobot/platforms/raspi"
)

func main() {
        r := raspi.NewAdaptor()
        servo := gpio.NewServoDriver(r, "24")

        work := func() {
		servo.Center()
		time.Sleep(1*time.Second)
		fmt.Println("moving servo to center")

		servo.Min()
		time.Sleep(1*time.Second)
		fmt.Println("moving servo to min")

		servo.Max()
		time.Sleep(1*time.Second)
		fmt.Println("moving servo to max")
        }

        robot := gobot.NewRobot("servoBot",
                []gobot.Connection{r},
                []gobot.Device{servo},
                work,
        )

        robot.Start()
}
```

a1xon, 2020-08-10:
https://raspberrypi.stackexchange.com/questions/70878/error-gpio-num-is-not-enabled-for-pi-blaster
You have to 
`sudo ./pi-blaster --gpio 13`

gen2thomas, 2023-11-10:
I'm going to close this issue as a duplicate, because it is now referenced by an issue which collect all the various problems with raspi and gpio.servo.

closed 2023-11-10 by gen2thomas