hybridgroup/gobot #759: Adafruit DC Hat Driver wrongfully assumes both the Servo and DC Motors Hats are present

crazyquark, 2020-07-06:
In this code:
https://github.com/hybridgroup/gobot/blob/master/drivers/i2c/adafruit_driver.go#L152
If the Servo HAT is not installed, it will not return without trying to connect to the DC Motors HAT. 
This makes it impossible to use the Adafruit DC Motors HAT(https://www.adafruit.com/product/2348): 

I think if the connection to the Servo HAT fails(which is this product https://www.adafruit.com/product/2327) it should be just set to nil and then try to connect to the motors HAT.

gandaldf, 2021-02-05:
Can you please check if my proposed solution #783 works for you?

gen2thomas, 2022-08-25:
@deadprogram it seems this fix is merged already, can we close the Issue?

closed 2022-09-01 by gen2thomas