hybridgroup/gobot #763: Bebop2 battery level

marcodiiga, 2020-08-29:
I'm using the bebop driver and adaptor and I couldn't find a way to receive a battery level event, something like

```go
drone.On(drone.BatteryLevel, func(data interface{}) {
	log.Printf("battery level: %d\n", data)
})
```
similar to the minidrone driver.

Is this missing or am I just overlooking something trivial?

gen2thomas, 2022-09-20:
@marcodiiga I think there is nothing implemented yet for the bebop.

gen2thomas, 2024-02-10:
@marcodiiga after get a deeper look in the BLE adapter and driver implementation I can provide the following information (provided names are based on current dev branch state).

The parrot minidrone has implemented a characteristic `batteryChara      = "9a66fb0f0800919111e4012d1540cb8e"` with a subscription to that and an event `parrot.BatteryEvent`. This implementation is not available for parrot bebop. The parrot bebop implementation is TCP based, not BLE, so there is no fast way to take over the existing implementation.

If there is no objection, I will close this issue in the next few days.

closed 2024-11-01 by gen2thomas