hybridgroup/gobot #765: can not run the code twice because of this ble error

HasanFarag, 2020-09-19:
when I run sudo ./minidrone D0:3A:7F:0D:E6:21 on ubuntu 20.04 i got that error if i want to run the code again after landing
```
2020/09/19 17:36:29 Initializing connections...
2020/09/19 17:36:29 Initializing connection BLEClient-32A33693B9E2A9F9 ...
2020/09/19 17:36:29 Initializing devices...
2020/09/19 17:36:29 Initializing device Minidrone-5CF68D44CFC83AD2 ...
2020/09/19 17:36:29 Robot minidrone initialized.
2020/09/19 17:36:29 Starting Robot minidrone ...
2020/09/19 17:36:29 Starting connections...
2020/09/19 17:36:29 Starting connection BLEClient-32A33693B9E2A9F9...
2020/09/19 17:36:39 1 error occurred:
	* can't connect to device default: can't get device: can't init hci: no devices available: (hci0: can't up device: connection timed out)
```

gen2thomas, 2024-02-10:
Hi @HasanFarag , since 2020 the used library has changed a bit. Additionally some things has changed recently in adapator and driver on dev branch. If this is still of interest, please can you repeat your test with the current implementation on dev branch and provide your running example. 

Furthermore I can see missing usage of error return values in gobot examples. I will fix this.

gen2thomas, 2024-11-05:
Fixed with [v2.4.0](https://github.com/hybridgroup/gobot/releases/tag/v2.4.0)

closed 2024-11-05 by gen2thomas