hybridgroup/gobot #772: Support for Jetson Nano platform

soerenbrockmann, 2020-12-17:
Do you plan to support Jetson Nano platform which runs on Ubuntu?
They mention that their platform (GPIO Pins) is equal to Raspberry PI. They only provide a Python :-(( GPIO Library.

deadprogram, 2020-12-17:
Yes, absolutely planning on this.

soerenbrockmann, 2020-12-17:
That would be amazing :-)

soerenbrockmann, 2020-12-17:
I just tested this example here:

`
package main

import (
        "gobot.io/x/gobot/platforms/raspi"
)

func main() {
		r := raspi.NewAdaptor()
		fmt.Println(r)
}
`

I got this value returned:

&{0x400001c358 RaspberryPi-62F43657CD24D5FD  map[] map[] 0 [<nil> <nil>] 0 0 [<nil> <nil>] 0 0 10000000}

Is it a good sign? 

Is there a way to test connectivity without using a pin (I am afraid of breaking my board :-) )

dlstjq7685, 2021-08-01:
@soerenbrockmann  
i just add feature jetson nano Adaptor.
check this branch.
https://github.com/hybridgroup/gobot/pull/799

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas