hybridgroup/gobot #774: MPU gyro freezes after a few seconds

willmendil, 2020-12-21:
High, 

I got a raspberry 4 with an arduino Due, 

I tried the MPU6050 code running with i2C, running this example https://gobot.io/documentation/examples/firmata_mpu6050/

I get some readings but after a few seconds it freezes and I need to restart the script. It feels like some buffer is filling up, but I'm not sure. The only main difference is that i'm using the MPU6500 instead of the 6050. 

After some digging, it looks like it freezes when it's trying to read line 101

```
...
100	data := make([]byte, 14)
101	_, err = h.connection.Read(data)
102	if err != nil {
...
```

in https://github.com/hybridgroup/gobot/blob/master/drivers/i2c/mpu6050_driver.go

Any idea? thanks

gen2thomas, 2022-09-20:
Hi @willmendil , possibly this is caused by a wrong initialization, see #838 .

Please could you test your device with the code of the provided PR #839 ?
Unfortunately I have only Tinkerboard and Digispark adaptors and use the MPU6050, so I can not say anything about your constellation.

gen2thomas, 2022-10-03:
MPU6050 is working with #839 on Digispark and Tinkerboard without freezing. So I'm going to close this issue, because it relates to MPU6500.

closed 2022-10-03 by gen2thomas