hybridgroup/gobot #784: Still have this! "ndefined: uuid.V2"

lloyd-g, 2021-02-10:
This issue was closed https://github.com/hybridgroup/gobot/issues/777
but I still get this
$GOARM=6 GOARCH=arm GOOS=linux go build go/src/gobot.io/x/gobot/examples/raspi_blink.go
# github.com/gobuffalo/uuid
go/src/github.com/gobuffalo/uuid/generator.go:32:13: undefined: uuid.NewV2
go/src/github.com/gobuffalo/uuid/uuid.go:41:7: undefined: uuid.V2

if you trace it down 
uuid.V2 is not defined in that package 

line 41 of go/src/github.com/gobuffalo/uuid/uuid.go:41:7
reads
	V2 = guuid.V2
not  uuid.V2
does this help?

I'm using 
go version
go version go1.15.7 linux/arm
on raspberry pi 3
but get same on 
Debian 10 with go1.15.7

deadprogram, 2021-02-10:
No release has gone out yet. That is why. Thanks for your patience. In the meantime, you can use the `dev` branch.

rafi0101, 2021-03-29:
Hi @deadprogram I have the same issue. When do you think you will release the next version?

jim-at-jibba, 2021-04-06:
Im new to go and so not sure how to use the dev branch of a package. 

Should I clone it and build? 

I tried to `go get` a specific branch but getting errors about the name of the package

```
go: github.com/hybridgroup/gobot dev => v1.15.1-0.20210127175256-97d23a5aec43
go get: github.com/hybridgroup/gobot@v1.15.1-0.20210127175256-97d23a5aec43: parsing go.mod:
        module declares its path as: gobot.io/x/gobot
                but was required as: github.com/hybridgroup/gobot
```

Thanks for the help

rafi0101, 2021-04-06:
Hi @jim-at-jibba it is very easy, once you know it :)
You go to your GOPATH (usually the go folder in your home directory). Then go to the directory: `go/src/github.com/gobuffalo/uuid/` and now you are in the git repository and can change the branch with `git checkout dev`

jim-at-jibba, 2021-04-06:
> Hi @jim-at-jibba it is very easy, once you know it :)
> You go to your GOPATH (usually the go folder in your home directory). Then go to the directory: `go/src/github.com/gobuffalo/uuid/` and now you are in the git repository and can change the branch with `git checkout dev`

Ha, I feel silly now! Thanks for your help

peterkingsbury, 2021-04-07:
Followed these steps:

```
go get -d -u gobot.io/x/gobot/...
go build
(got the gobuffalo uuid.NewV2 errors)
(google'd, found the closed issue, searched the issue history, ended up here)
cd $GOPATH/src/github.com/gobuffalo/uuid/
git status
> On branch master, etc
git checkout dev
> error: pathspec 'dev' did not match any file(s) known to git
git fetch
git branch -a
> * master
>   remotes/origin/HEAD -> origin/master
>   remotes/origin/master
```
Any help would be appreciated! Many thanks :-)

gen2thomas, 2021-04-07:
Hi @deadprogram you mean with...
> No release has gone out yet. That is why. Thanks for your patience. In the meantime, you can use the `dev` branch.

... the dev branch of this repo (gobot)?

Hi @peterkingsbury , @jim-at-jibba I have made my local change in $GOPATH/src/github.com/gobuffalo/uuid/ folder (on master branch without commit and push) to make it work like this:

```
diff --git a/generator.go b/generator.go
index 886d39c..718e785 100644
--- a/generator.go
+++ b/generator.go
@@ -29,7 +29,7 @@ import (
 var NewV1 = guuid.NewV1
 
 // NewV2 returns DCE Security UUID based on POSIX UID/GID.
-var NewV2 = guuid.NewV2
+//var NewV2 = guuid.NewV2
 
 // NewV3 returns UUID based on MD5 hash of namespace UUID and name.
 var NewV3 = guuid.NewV3
diff --git a/uuid.go b/uuid.go
index da62416..d01a838 100644
--- a/uuid.go
+++ b/uuid.go
@@ -38,7 +38,7 @@ type UUID = guuid.UUID
 // UUID versions
 const (
        V1 = guuid.V1
-       V2 = guuid.V2
+       //V2 = guuid.V2
        V3 = guuid.V3
        V4 = guuid.V4
        V5 = guuid.V5

```

benhadad, 2021-05-19:
I am new to Gobot, but this issue makes me feel I am starting out on a dead pathway with a project that isn't being fully supported anymore.   Last update was 6 months ago and didn't address this issue?

cwinters8, 2021-09-29:
For posterity, I was able to build successfully after doing the following to get the dev branch updates for uuid.

```sh
go get gobot.io/x/gobot@dev
```

charliemcelfresh, 2022-06-19:
This worked for me : `go get -u github.com/gobuffalo/validate`

gen2thomas, 2022-09-01:
@deadprogram because the dev branch was merged for v1.16.0 in May 2022, is this bug now resolved?

deadprogram, 2022-09-11:
Oops, forgot to close. Yes. :smile_cat:

closed 2022-09-11 by deadprogram