hybridgroup/gobot #792: No mac M1 support?

cyantarek, 2021-05-01:
I'm getting this error while running on my new apple silicon macbook M1.

```
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:263:18: undefined: baudrateMap
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:268:23: undefined: baudrateMap
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:273:20: undefined: toTermiosSpeedType
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:274:20: undefined: toTermiosSpeedType
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:283:22: undefined: tcCMSPAR
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:288:22: undefined: tcCMSPAR
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:293:22: undefined: tcCMSPAR
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:296:6: undefined: tcCMSPAR
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:301:21: undefined: tcCMSPAR
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:304:6: undefined: tcCMSPAR
../../go/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:304:6: too many errors
```

deadprogram, 2021-05-01:
Needs an update to the latest version of the `go.bug.st/serial` package.

A PR adding this would be gladly accepted! :)

electric-pony, 2022-12-30:
Looks like this is still an open issue. I have the same problem on my M1 mac.

Avaloss22, 2023-01-11:
> Looks like this is still an open issue. I have the same problem on my M1 mac.

Did you solve it? I have the same problem :/

gen2thomas, 2023-01-13:
Do you have tested with dev or release branch?

gen2thomas, 2023-05-19:
Hi @rosricard , @Avaloss22 , @cyantarek we have released v2.0.0 with some updated modules. Please give some feedback if this fixes your issues.

Thanks!

rosricard, 2023-05-19:
Great! I should be able to take a look at this in the next few days.

gen2thomas, 2023-06-19:
Hi, are there any news on this topic?

gen2thomas, 2023-11-10:
I'm going to close this issue for now. Please feel free to reopen, if the problem still exist.

closed 2023-11-10 by gen2thomas