hybridgroup/gobot #798: Tello driver FlightData fields description

orsinium, 2021-06-15:
The `FlightData` struct fields use a lot of abbreviations and don't have descriptions. What is `EmOpen`? `ElectricalMachineryState`? `FlyMode`? `FrontLSC`? `ImuState`? What are the possible values for `CameraState`? Is there a place where I can read about it?

orsinium, 2021-06-15:
Found about IMU: [Ryze Tello IMU Calibration](https://droneguides.net/en/ryze-tello-imu-calibration/). So I guess `ImuState` says if the drone needs IMU calibration. It's still not clear what are the values for `ImuCalibrationState`, though.

orsinium, 2021-07-02:
Tello SDK (Tello EDU only?) has a description for the state fields:
https://dl-cdn.ryzerobotics.com/downloads/Tello/Tello%20SDK%202.0%20User%20Guide.pdf

It is different from what we have but there are a few matching fields.

orsinium, 2021-07-02:
`FlightData.LightStrength` field appears to be unused 🤔

gen2thomas, 2022-09-20:
Hi @orsinium , I'm not a user of this device, so I can not tell something about the structure elements. Because it seems you are (were) a novice for using this driver you are best suited for adding some comments for the next new user.  

So, please could you provide a PR with some comments besides the elements or at least provide a complete list here so anyone else can prepare an PR? This would be perfect.

orsinium, 2022-09-20:
I don't remember much at this point, only what I wrote in this issue. But I think I haven't found anything else. 

I'm pretty sure I'm not the first one to use the driver. I vaguely remember @deadprogram showing people how to fly a DJI Tello drone at some conferences. And then there is [this article](https://muetsch.io/flying-a-dji-tello-drone-with-go.html). So, I hope there are some people in the world who can uncover the mysteries of the fields I didn't uncover.

I'll try to find some time in a week and contribute comments to the fields that I described in this thread. The rest I don't know yet.

gen2thomas, 2022-09-20:
Thanks, that would be great!

PR #864 (2022-10-03): Document fields for tello.FlightData
1. Document some of the fields in tello.FlightData description of which I found in my old notes.
2. Drop LightStrength and SmartVideoExitMode. These fields are always zero in the current implementation.

Closes #798