hybridgroup/gobot #802: Raspberry pi PICO

hajsf, 2021-08-30:
Is gobot working with the new raspberry pi pico controller, different in the setup required from running raspberry pi with gobot, or every thing is the same as both having almost same processor?

https://www.raspberrypi.org/documentation/rp2040/getting-started/

gen2thomas, 2022-09-20:
Hi @hasanAjsf , it seems there is no information available at this point. In the meantime, do you have made some experiences with the PICO, which you can share with us?

jackjameshoward, 2023-01-09:
has there been any more consideration given for support to this device? 
I don't know how similar gobot is to tinygo, but there is support for the pico over there.   
https://tinygo.org/docs/reference/microcontrollers/pico/

gen2thomas, 2023-01-13:
As far as I can tell Microcontrollers are supported by [firmata](https://github.com/hybridgroup/gobot/tree/release/platforms/firmata) protocol in gobot. So, if the firmata firmware supports the hardware it should work, but I have no experience with that. The "normal [Raspi](https://github.com/hybridgroup/gobot/tree/release/platforms/raspi)", which is supported by gobot, is a computer with installed Linux OS.

gen2thomas, 2023-07-07:
I'm going to close this issue, because there are 2 alternative ways available to use the PICO microcontroller. @hasanAjsf feel free to reopen the issue with providing more information why you need/want to use gobot over tinygo.

closed 2023-07-07 by gen2thomas