hybridgroup/gobot #815: return int in AnalogSensorDriver.Read() prevents nice implementation

gen2thomas, 2022-03-30:
I'm currently implementing an generic temperature sensor driver for linear scaling (PT100, NTC) and NTC scaling.
IMO the best way would be wrapping the AnalogSensorDriver, but the return value is integer, which not really fits the need of an analog reading in my opinion. My suggestion is to change the interface to float64 and adjust all the examples (I would and can do that).

Another way around would be extend the driver by a new function "ScaledRead() float64" and some unattractive adjustments inside the driver.

@deadprogram what do you think?

deadprogram, 2022-04-09:
Hmm, it is a breaking change, but not a very large one.

I'd say, let's go for it unless anyone else objects.

gen2thomas, 2022-04-24:
I've completely forgotten this question and your answer and to make some changes to #814 before merging (which the advantage that is currently backward compatible). But I think, now we can see the "unattractive adjustments" I mentioned in the "analog_sensor_driver.go". There are two new functions named ReadValue() and Value() which the name "value" is meaningless in this context.

I will prepare a PR in the next few days to improve this according to your answer and I hope we can see, that the result is more readable. This has also the advantage to see the difference more clearly. If the improvement does not outweigh the disadvantage of the small breaking change, we can just drop the PR and close this issue.

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas