hybridgroup/gobot #816: Does this Lib. Still Work w/ the BeagleBone Black?

silver2row, 2022-04-08:
Hello,

I have not tried this out yet. I plan on trying it out to handle some LED to start via the BeagleBone Black and RelayCape.

Here: https://github.com/hybridgroup/gobot/tree/master/platforms/beaglebone

Is this still current or would I need to update in my cloned repo. a way to handle this lib?

Seth

gen2thomas, 2023-01-23:
Hi @silver2row have you been able to check this out yet ? Could you test the dev branch for your requirements?

stevstrong, 2023-10-18:
The newest package does have a problem at install.
I followed this page:
https://gobot.io/documentation/platforms/beaglebone/
Having Go already installed, I started to install gobot with:
`go get -d -u gobot.io/x/gobot/...`
And after downloading a lot f packages, it throws these lines:
```
gobot.io/x/gobot/drivers/spi imports
        periph.io/x/periph/conn/physic: cannot find module providing package periph.io/x/periph/conn/physic
gobot.io/x/gobot/drivers/spi imports
        periph.io/x/periph/conn/spi: cannot find module providing package periph.io/x/periph/conn/spi
gobot.io/x/gobot/drivers/spi imports
        periph.io/x/periph/host/sysfs: cannot find module providing package periph.io/x/periph/host/sysfs
```
So I assume something is wrong there.
Can someone help to fix this?

gen2thomas, 2023-10-18:
Hi @stevstrong , the documentation at [this page](https://gobot.io/documentation/platforms/beaglebone/) is outdated. We only support the v2 of gobot. Please have a look at this example instead: https://github.com/hybridgroup/gobot/blob/release/examples/beaglebone_blink.go.

gen2thomas, 2023-10-18:
This depends on your start point. If you already have a go program or project for your platform a `go mod tidy` should be sufficient (after you have changed the old imports to "gobot/v2"). Otherwise, please read the guide and try it step by step.

gen2thomas, 2024-11-01:
Hi @stevstrong , do you have any progress on that? Can we close this issue?

Thanks Thomas

stevstrong, 2024-11-02:
Hi,
 I abandoned this project, so feel free to close this if you think that it is not relevant anymore.

gen2thomas, 2024-11-05:
Hi @stevstrong , thanks for the fast response, so I will close this issue.

closed 2024-11-05 by gen2thomas