hybridgroup/gobot #827: ads1x15 driver not working stable when reading multiple inputs

gen2thomas, 2022-08-25:
Currently I'm testing a module with ADS1115 and 2 used analog input pins. Most of the time the values of both inputs are converted as the same, although the voltage differs (measured by a meter).

The main reason seems to be the wait [line 350 in driver code](https://github.com/hybridgroup/gobot/blob/d8081f912d324f570c6e3fa74aba7c203e53bb27/drivers/i2c/ads1x15_driver.go#L350) which tries to delay, but is to small or at the wrong position.

According to the datasheet of the ADS111x in the table "Config Register Field Descriptions" the "OS can only be written
when in power-down state and **has no effect when a conversion is ongoing.**". A raw implementation of a "wait for bit15 is 1" seems to fix the problem.

I will make further tests and can possibly provide a fix by one of the both approaches or add a With...() function for that.

gen2thomas, 2022-08-25:
Some further investigations shows that the implemented delay is quite correct (7.912ms for 128 SPS). According to the datasheet the device is compatible with standard, fast and high-speed mode, so this should be no cause for the problem. In the section "Single-Shot Mode", the wake-up time should be fast enough "When the OS bit is asserted, the
device powers up in approximately **25 μs**, resets the OS bit to 0, and starts a single conversion.", so this should also be no cause of the problem. 

My device needs arround 32ms for one conversion. Possibly the purchased device is of low quality and don't fulfill the specifications.

So I will go with the safest approach and implement the code like in python. By adding a With...() function, this default behavior can be disabled.

deadprogram, 2022-08-30:
Sounds great @gen2thomas thanks!

gen2thomas, 2023-05-15:
part of release v2.0.0

PR #829 (2022-10-11): BUGFIX/IMPROVEMENTS: ADS1x15
* fix #827 
* work with stable behavior by default (maybe slower)
* introduce a With function to switch to old, maybe the faster approach
* include improvements for #828
closed 2023-05-15 by gen2thomas