hybridgroup/gobot #830: HMC5883L driver returns wrong values

gen2thomas, 2022-08-31:
According to the datasheet section "Data Output X Registers A and B" the values should be in 2s complement form +0xF800 ... 0x07FF (-2048 ... +2047). In case of an overflow, the value gets "-4096 (0xF000)".

In my first test case, the values of X,Y, Z jump in a high range. All values can exceed 2048 and don't lock to -4096 for a strong magnetic field.

A first investigation shows, that a missing byte-swap after reading the MSB, LSB data with "ReadWordData()" could be the root cause.

I will provide a fix after some further tests, maybe together with #805 .

gen2thomas, 2022-09-01:
After reading the [SMBus specification](http://www.smbus.org/specs/SMBus_3_2_20220112.pdf), I come to the conclusion, that the order is defined as LSB, MSB. This means the direct usage of the SMBus access in sysfs/i2c_device leads to the wrong order for I2C usage for my Tinkerboard in combination with the HMC5883L driver (without swap).

gen2thomas, 2022-09-03:
I have tested the HMC5883L driver today with my digispark platform and the same problem happen. So I looked again to the digispark driver implementation, and I was wrong with my statement that the bytes become swapped, because it reads 2 bytes and the first becomes the low byte, the second the high byte. I will correct my comments above.

gen2thomas, 2022-09-03:
To clarify, how a word read "normally" works, I installed my mcp2221 USB-to-UART/I2C serial converter and connected the HMC5883L to it. Than I initializes the device and read the data for X in different ways:
* i2cget -y 7 0x1e 0x03 --> 0xff
* i2cget -y 7 0x1e 0x04 --> 0xfd
* i2cget -y 7 0x1e 0x03 w --> 0xfdff

This means, that also the commonly used i2cget binary will return a word as LSB, MSB, which matches the implementation of all our platforms. So the HMC5883L driver (and maybe the two others) needs to be fixed.

deadprogram, 2022-09-06:
Looks like something must have gotten mixed up at some point. Thanks for finding this issue!

gen2thomas, 2023-05-15:
part of release v2.0.0

PR #831 (2022-10-03): BUGFIX: missing HMC5883L byte swap
* fix #830 
* add unit tests for reading
* add some constants and With functions for "samples averaged", "data output rate", "applied bias" and "gain"
* fix #805
* use ReadBlockData()
* i2c.Driver used as base, which let us more focus on important tests
closed 2023-05-15 by gen2thomas