hybridgroup/gobot #832: i2c/smBus ReadBlockData() not implemented

gen2thomas, 2022-09-04:
According to the [Kernel specification](https://docs.kernel.org/i2c/smbus-protocol.html#smbus-block-read) it should be possible to implement this function. The opposite ["WriteBlockData()" is still implemented](https://github.com/hybridgroup/gobot/blob/d8081f912d324f570c6e3fa74aba7c203e53bb27/drivers/i2c/i2c.go#L44).

Following drivers could be simplified:
* [th02_driver.go](https://github.com/hybridgroup/gobot/blob/d8081f912d324f570c6e3fa74aba7c203e53bb27/drivers/i2c/th02_driver.go#L234)
* [mpu6050_driver.go](https://github.com/hybridgroup/gobot/blob/d8081f912d324f570c6e3fa74aba7c203e53bb27/drivers/i2c/mpu6050_driver.go#L100)
* [mpl115a2_driver.go](https://github.com/hybridgroup/gobot/blob/d8081f912d324f570c6e3fa74aba7c203e53bb27/drivers/i2c/mpl115a2_driver.go#L120) and for [getData()](https://github.com/hybridgroup/gobot/blob/d8081f912d324f570c6e3fa74aba7c203e53bb27/drivers/i2c/mpl115a2_driver.go#L156)
* [l3gd20h_driver.go](https://github.com/hybridgroup/gobot/blob/d8081f912d324f570c6e3fa74aba7c203e53bb27/drivers/i2c/l3gd20h_driver.go#L115)
* HMC5883L in ReadRawData()
* [ccs811_driver.go](https://github.com/hybridgroup/gobot/blob/d8081f912d324f570c6e3fa74aba7c203e53bb27/drivers/i2c/ccs811_driver.go#L339), an implementation of read block data, also used in all 3 bmpx80 and bme280 drivers

deadprogram, 2022-09-06:
That sounds like a great idea!

gen2thomas, 2022-10-01:
While rework of PCA9501 (see #786 ), I found that all implementations of Read*Data() in digispark (and possibly also in firmata) are wrong implemented, regarding to the [Kernel specification](https://docs.kernel.org/i2c/smbus-protocol.htm) and the implementation of sys-fs driver.

[ReadByteData()](https://docs.kernel.org/i2c/smbus-protocol.html#smbus-read-byte) (Comm==Register)
S Addr Wr [A] Comm [A] **Sr** Addr Rd [A] [Data] NA P
We have implemented:
S Addr Wr [A] Comm [A] **P S** Addr Rd [A] [Data] NA P

In general most chips support both sequences, but unfortunately not all, e.g. PCA9501, which only support the "Read Byte".

gen2thomas, 2022-10-01:
Hi @deadprogram , because I'm not familiar with firmata in general and with client "writes an arbitrary Sysex command to the microcontroller" in special, I'm just marked the wrong implemented commands with an TODO and will create a new issue for this special behavior.

gen2thomas, 2023-05-15:
part of release v2.0.0