hybridgroup/gobot #834: How to set pin is pulled high or pulled low or not pulled?

hacktronics, 2022-09-12:
How to set pin is pulled high or pulled low or not pulled? The question is specific to Raspberry Pi, but in general I couldn't find in the button driver, how to set the pin is pulled or not.
In one support question, I found how to change the default push state, but does gobot support pull characteristics of a pin?

gen2thomas, 2022-09-12:
Please could you specify which driver you use. Maybe I could help, although I have no Raspi but Tinkerboard, most parts are implemented identically.

hacktronics, 2022-09-20:
I mean for buttons, how do you set the pulled mode in the Raspberry Pi.
https://gobot.io/documentation/examples/raspi_button/
I found how to change the default push state, but does gobot support pull characteristics of a pin?

gen2thomas, 2022-09-20:
Possibly you mean a pullup or pulldown resistor to prevent floating input, e.g. for button or an open collector (e.g. when using opto couplers). Not all devices have internal resistors and some, which have them, are not all supported by gobot. So the best way is to apply an external resistor.

In case you have an pulldown resistor and an button to positive voltage, this will just cause inverse logic.

Please also have in mind that some debouncing logic should be added in case you will count something or pulses are not allowed for your actuator. Maybe this is good enough filtered by the 10ms poll interval, but this depends on your special hardware.

gen2thomas, 2023-01-23:
the internal pull up/down feature will be implemented with #897

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas