hybridgroup/gobot #835: How to dispose a pin?

hacktronics, 2022-09-12:
I have created a leddriver for a pin, let say its no longer needed. In that case, how to dispose that pin and free resources, so that I can make that pin as input button?

gen2thomas, 2022-09-12:
Which adapter do you use and which driver?

Normally you can just switch the pin from output to input by using the related function. But which use case do you have for such an on-the-fly switching?

hacktronics, 2022-09-20:
I am building a MIT Scratch based block based coding software for kids. So in https://ide.codeskool.cc/, this kids can assign a pin and perform some task. Now its hard to stop them from assigning the pin back as a button from a LED. So, for example in https://gobot.io/documentation/examples/raspi_button/, if the pin is previously a LEDDriver, how to dispose that pin and create a new button driver.

gen2thomas, 2022-09-20:
Nice project!

IMO this is a special use case for simulation / code generation only. In reality the hardware is plugged or soldered together, which means a pin is not changeable during the software is running. Therefor also a pin direction will not change. This will be only happen if the hardware will be reworked.

When the count of GPIOs are not enough, and therefore a pin should be used for more than one In/Out, using a multiplexer or an bus (i2c) would be my choice.

Possibly this practical idea behind should be also part of your project? This means, if a pin is added/removed/changed, the system needs to be stopped and initialized afterwards (maybe in the background for easy user experience). Is there a gobot code generator in the background?

hacktronics, 2022-09-21:
https://youtu.be/Kp_Isk_9Mu8, the gobot is an executable running on the Raspberry Pi, its not a code generator. For kids, they won't be writing code or compiling it. So, the executable would be running just once, hence I don't want the system/exe to be stopped and rerun again when the user make a button from a pin, that was earlier a LED. To keep it simple, there is no hardware shield, we are not selling hardware like SunFounder, etc. its a free solution. So the idea of multiplexer won't work here.
So, from this discussion, if something breaks, for example on pin switching, the only way is for the user to rerun the program, which would remove all previous assignments. But I thought a Pin dispose method would be good and wouldn't require shutting program done and running it again and reconnecting to CodeSkool.

gen2thomas, 2022-09-21:
Thanks for the information. Please could you provide your fixed gobot code to have a look, how a button or an LED is defined at first time. What will happen, when a kid change the pin of LED (without changing the direction) - is this currently working - how the code handle this? Do you have already tested a Halt-ChangeType-Start - sequence?

gen2thomas, 2023-06-16:
Hi @hacktronics could you find a solution for your problem? Maybe it is possible now by using the gpiod (cdev) based implementation on your hardware.

gen2thomas, 2023-07-07:
I close this issue due to no activity since some month. Feel free to reopen if this is still a question.

closed 2023-07-07 by gen2thomas