hybridgroup/gobot #848: go run error - go version go1.19.1 darwin/amd64

mstrYoda, 2022-09-22:
```bash
❯ go run main.go
# golang.org/x/sys/unix
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/syscall_darwin.1_13.go:29:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.1_13.go:27:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.1_13.go:40:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.go:28:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.go:43:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.go:59:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.go:75:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.go:90:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.go:105:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.go:121:3: //go:linkname must refer to declared function or variable
../../pkg/mod/golang.org/x/sys@v0.0.0-20201126233918-771906719818/unix/zsyscall_darwin_amd64.go:121:3: too many errors
```

gen2thomas, 2023-07-07:
Hi @mstrYoda , please can you provide some more information of your main program and which example or driver you try to implement?

mstrYoda, 2023-07-07:
> Hi @mstrYoda , please can you provide some more information of your main program and which example or driver you try to implement?

Hi @gen2thomas, I am so sorry I do not know how I created an issue like this without any context of further information.

I forgot the case in here so may be it is better to close this issue.

closed 2023-07-07 by mstrYoda

gen2thomas, 2023-07-07:
@mstrYoda thanks for your fast response!

Best regards, Thomas