hybridgroup/gobot #849: CI/CD environment outdated

gen2thomas, 2022-09-24:
* deprecated  (circleci), see https://discuss.circleci.com/t/legacy-convenience-image-deprecation/41034
* golang [1.15 is used](https://github.com/hybridgroup/gobot/blob/f141a2bc7a562f500ced5ab679bede9816fa01f1/.circleci/config.yml#L6), which cause CI build errors for new functions of golang 1.16 (e.g. os.ReadDir)

deadprogram, 2022-09-24:
Fixed!

gen2thomas, 2022-09-25:
Mhm, unfortunately the change from /go to $HOME/go in new cimg seems to make some problems, see
https://app.circleci.com/pipelines/github/hybridgroup/gobot/41/workflows/dfaa8bbf-48ed-4b6e-b91b-bbceedd52bba/jobs/41?invite=true#step-101-73

gen2thomas, 2022-09-25:
After merging the last fix, tests will be start. But following tests now fail:
* "utils_test.go:38: Every was not called"
* "utils_test.go:44: Every should have stopped"

Because this tests running before without any problems, it seems to be flaky.

Also this test seems to be flaky:
*  "client_test.go:466: SysexResponse was not published"

deadprogram, 2022-09-25:
Yes, they appear to be very sensitive to race conditions/timing.

gen2thomas, 2022-09-27:
still unstable:
https://app.circleci.com/pipelines/github/hybridgroup/gobot/68/workflows/9f80bfb8-d832-4e19-8c26-262ce31b116c/jobs/75

gen2thomas, 2023-05-15:
part of release v2.0.0

PR #850 (2022-09-24): CI/CD switch to golang 1.19 and to new cimg
fix #849