hybridgroup/gobot #852: Makefile outdated

gen2thomas, 2022-09-25:
The "fmt_check" and "test_with_coverage" will not work in general, because the folder ".ci" is not exist. Also the excluding of vendor folder for tests should normally not be needed anymore.

suggested change for "fmt_check":
* change name to "style"
* call "go fmt ./..."
* some ideas for deprecated/frozen "golint" are needed

suggested change for "test_with_coverage":
* call "go test -v ./... -coverprofile=coverage.txt" (adding "coverage.txt" to .gitignore)
* call or suggest call "go tool cover -html=coverage.txt # will open the coverage in browser"

https://stackoverflow.com/questions/10516662/how-to-measure-test-coverage-in-go

After everything is working again, this could be mentioned in "CONTRIBUTING.md".

Alternatively, if nobody needs the "Makefile", it can be removed completely.

gen2thomas, 2022-09-28:
Hi @deadprogram , after the main problems are fixed, how want we proceed?
* Do we want to include examples build in circleci?
* Do we want to use the Makefile for circleci build ("make test", "make examples" "make fmt_check" etc.)? Or do you prefer to program this code parts separately in a command-section of config.yml?
* If we keep the Makefile (IMO this is nice for golang newbies), do we want to reference the usage from CONTRIBUTING.md?
* How do we proceed with the formatting stuff? E.g. since go 1.17 the build tags have changed to "//go:build" and the formatter ("make fmt_fix") will change all examples and a few more files.

gen2thomas, 2022-09-28:
Maybe useful to make a decision:
* adaptor test takes ~20s
* driver tests takes ~25s
* examples build takes ~1:45

deadprogram, 2022-09-28:
OK then not examples :)

gen2thomas, 2022-10-20:
Hi @deadprogram , check examples with "go vet" takes only ~50sec in circleci. Should we add it, what do you think?

deadprogram, 2022-10-23:
> check examples with "go vet" takes only ~50sec in circleci.

Probably a good idea, keep them from getting messy. :smile_cat:

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas