hybridgroup/gobot #860: go.mod and some dependencies outdated

gen2thomas, 2022-09-27:
currently go 1.13 is used in go.mod
and also some used dependencies can switch to a newer version

this guides are very useful: 
* https://go.dev/doc/modules/managing-dependencies
* https://golang.cafe/blog/how-to-upgrade-golang-dependencies.html

long story short:
go list -m -u all
go get -t -u ./...
go mod tidy
make test

deadprogram, 2022-09-27:
See https://github.com/hybridgroup/gobot/pull/856 where I have updated a bunch f these already.

gen2thomas, 2022-09-27:
Hi @deadprogram I have seen that you have already started upgrading with #856 .

Is the go version 1.16 used intentionally?
Debian bullseye uses 1.15, backports 1.19 and bookworm will most likely use 1.19.

deadprogram, 2022-09-27:
Actually I meant to pick 1.17 with intention like Go to support current version minus 2.

gen2thomas, 2022-09-27:
Ok, so we should align it with used version in circleci? Currently I have chosen "cimg/go:1.19".

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas