hybridgroup/gobot #865: Driver for HC-SR04

Juan1ll0, 2022-10-16:
Hi guys,
  I'm development and IOT device that need to measure a water tank level. I'm think that using an HC-SR04 is a good option but i find that Gobot doesn't have support for that sensor. I found library at [](https://github.com/raspberrypi-go-drivers/hcsr04) that works fine. I know that this sensor can not be used under "firmata" adaptor because of pulseIn function is not implemented, but with "raspi" Can I found any solution?.

I try to create a new Driver adopting code of [](https://github.com/raspberrypi-go-drivers/hcsr04) to gobot using a DirectPInDriver but doesn't work.

gen2thomas, 2022-10-19:
Like already mentioned, firmata is not best suitable because "pulsIn". I just add some issues as reference:
* #152 
* #246 
* #361 
* [firmata PR](https://github.com/firmata/arduino/pull/396)

gen2thomas, 2022-10-20:
Hi @Juan1ll0 , please could you provide your new driver (with an example) by a Draft-PR to dev-branch. So maybe we can find a solution.

Juan1ll0, 2022-10-21:
> Hi @Juan1ll0 , please could you provide your new driver (with an example) by a Draft-PR to dev-branch. So maybe we can find a solution.

Juan1ll0, 2022-10-21:
Sorry it doesn't work the code only compiles. I try to transform code like structs, pins and similar but it doesn't work. For example the mentioned project used [github.com/stianeikeland/go-rpio/v4](url) and i transform pins to DirectPinDriver gobot implementation and similar things. If you want I can send you the code.

gen2thomas, 2022-10-21:
@Juan1ll0 please create a branch with your code and file a Draft-PR or add the files here (e.g. as zip) and I can do that.

Juan1ll0, 2022-10-21:
> 

OK. I'll do it

Juan1ll0, 2022-10-21:
> @Juan1ll0 please create a branch with your code and file a Draft-PR or add the files here (e.g. as zip) and I can do that.

You've got at [https://github.com/hybridgroup/gobot/pull/872](url). Sorry about my code but i'm not an experto on go. Than you by your interest.

gen2thomas, 2022-10-21:
Ok, could you please describe:
* which adaptor do you use and which specific hardware and installed OS/FW
* how is the setup (e.g. you could provide an example in your PR)
* what exactly does not work (Do you have a specific error, timeout etc?)

Juan1ll0, 2022-10-21:
> Ok, could you please describe:
> 
> * which adaptor do you use and which specific hardware and installed OS/FW
> * how is the setup (e.g. you could provide an example in your PR)
> * what exactly does not work (Do you have a specific error, timeout etc?)
I use a raspberry pi 4 with Raspian, with Gobot 1.16 and Go 1.18.
Adaptor: raspi
Setup: On pins 11 for trigger and 13 for echo
Error: Timeout - message: ERRO[0000] impossible to measure distance                error="echo not received after 60 milliseconds"

I thought the problem is on function StartDistanceMonitor or in getPinStateChangeTime

gen2thomas, 2022-10-23:
@Juan1ll0 please see my remarks in PR.

gen2thomas, 2023-10-30:
available since v2.2.0

PR #1012 (2023-10-27): gpio(hcsr04): add driver for ultrasonic ranging module
## Solved issues and/or description of the change

* continued work of @Juan1ll0, thanks for the great starting point
* makes #872 obsolete
* fix #865 
* depends on #1015 
* test for i2c removed from appveyor (windows environment), because i2c-gpio implementation depends on linux by build tag in gpiod
* tests for firmata-adaptor excluded from appveyor by build tag
* add "WithGpioPollForEdgeDetection" at adaptor level

## Manual test

- OS and Version (Win/Mac/Linux): Linux
- Adaptor(s) and/or driver(s): raspberry, tinkerboard

## Checklist [... cut]
closed 2023-10-30 by gen2thomas