hybridgroup/gobot #868: joystick：Compiling and running exceptions

wushu037, 2022-10-18:
Thanks a lot for providing the joystick library, it works fine on Mac, but there are exceptions on windows

I used the [sample code](https://github.com/hybridgroup/gobot/tree/release/platforms/joystick#examples) from joystick's readme to run

The first is a compilation error: joystick relies on [go-sdl2](https://github.com/veandco/go-sdl2), and this error occurs when it is compiled cross-platform from Mac to Windows and directly on Windows.
[image]

For this error. I raised an [issue](https://github.com/veandco/go-sdl2/issues/542) with go-sdl2, and later I replaced it with 0.4.25 as suggested, and it compiled successfully. So maybe joystick should update the version of go-sdl2

However, joystick is running into problems again: I'm using Xbox Wireless Controller (xsx2020) and it only occasionally listens to the trigger button, but not to all other keys!

I also tried to run scanner.go, but it quits immediately
[image]

gen2thomas, 2022-10-19:
Hi @wushu037, currently "github.com/veandco/go-sdl2 v0.3.3" is referenced by [go.mod](https://github.com/hybridgroup/gobot/blob/b285544e43d20e69d3b3ea19d9b236002b66006c/go.mod#L23). Your suggestion was to use "0.4.25". Therefor I assumed this means the module version, but maybe I'm wrong.

wushu037, 2022-10-20:
Hi @gen2thomas ,i read the issue you linked to, maybe I didn't understand it and didn't see any connection to the go-sdl2 module version.
but yes, go-sdl2 needs to be replaced with the latest version in order to compile properly on windows. Also, even if it runs successfully, there are issues to troubleshoot during runtime (see first comment)

gen2thomas, 2022-10-20:
Hi @wushu037 , I have now fixed the old modules on dev branch. I started "go run scanner.go" like you have done, but without any joystick attached and in a Linux box - it do not exit immediately, it waits. Unfortunately I have no windows system to test it.

What do you mean with "it works fine on Mac"? Do you have used the same example? I ask, because the example use a "dualshock3" type. Is this compatible with your "Xbox Wireless Controller (xsx2020)"?

wushu037, 2022-10-20:
Hi, @gen2thomas , thank you for your investigation of the problem
Here's the thing. I tested it on Mac and Windows, and it works fine on Mac, but on Windows it appears to be abnormal. I didn't test it on Linux

It is the example of the sample. I didn't give my program for the sake of describing the problem. Although the example uses "dualshock3", on Mac, my xsx2020 joystick still has multiple keys that can be listened to. But on Windows, only the trigger key is occasionally listened to

To better solve the problem, I give my program file. It implements a very simple function: mapping the joystick's share keys to the system's share keys. In this program, the configuration file is set up by me running scanner.go on my Mac, and also only the trigger key is responded to.

[归档.zip](https://github.com/hybridgroup/gobot/files/9828891/default.zip)

If you don't have Windows, you can install a virtual machine and the effect should be the same. Either on a win virtual machine or a win physical machine, I tried

gen2thomas, 2022-10-20:
Hi @wushu037 , I need to stop my support here, because windows and no joystick available. 

Please could you change the headline to which is more pointing to the remaining problem, e.g. this words/phrases should be included "windows, scanner.go, only one key". So someone can pick up the issue with existing hardware and deeper knowledge.

gen2thomas, 2024-11-01:
Hi @wushu037 , the sdl dependency is now replaced by another module. Maybe you can test again your device on Windows with the new implementation.

Thanks Thomas