hybridgroup/gobot #900: release request for gobot 2.0.0

gen2thomas, 2023-02-04:
Hi @deadprogram , a software is never perfect, but we get fixed a lot in the past month and furthermore, added some new features. What do you think about a gobot release? Can you do that?

Also the home page https://gobot.io/ was last updated some years ago and miss some already included drivers. Who is able to update this?

Thanks for your support, Thomas

wgroeneveld, 2023-02-09:
Hi, I'm contemplating on contributing to Gobot to add compatibility with a non-raspberry pi SBC, but looking at current pull requests and the latest checkin in the main branch, I wonder: is Gobot still actively being developed?

gen2thomas, 2023-02-09:
Hi, contributing is always welcome. The default branch is the "release" branch with latest commit in May 2022. Development effort is done on dev-branch, see [CONTRIBUTING.md](https://github.com/hybridgroup/gobot/blob/release/CONTRIBUTING.md).

gen2thomas, 2023-03-23:
Hi @deadprogram , maybe you are too busy to do the merge at the moment, maybe you want wait some time for any reason? Please let me know if I should do this and teach me how.

gen2thomas, 2023-04-01:
To answer my own question - the home page seems to be based on the resources of this repo: https://github.com/hybridgroup/gobot-site

deadprogram, 2023-04-01:
@gen2thomas the site is based on very old static site generator. Trying to get it working again to update is a seemingly big task. The best would be to migrate to Hugo?

gen2thomas, 2023-04-02:
Hi @deadprogram , I'm not familiar with Middleman or Hugo but RoR (long time ago), lets discuss this topic in the already existing https://github.com/hybridgroup/gobot-site/issues/61.

What do you think about a new release of gobot? Can you teach me about the steps to do?

gen2thomas, 2023-05-14:
I will do it in this way:
* prepare draft PR for code dev --> release
* prepare draft PR for CHANGELOG.md & version feature-branch --> dev
* merge code to release
* merge changelog to dev, afterwards to release
* create a new release with description from added changelog

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas