hybridgroup/gobot #968: cannot find modules on install

jney, 2023-07-03:
Install `go get -d -u gobot.io/x/gobot/...` terminate with the following errors

```
gobot.io/x/gobot/drivers/spi imports
	periph.io/x/periph/conn/physic: cannot find module providing package periph.io/x/periph/conn/physic
gobot.io/x/gobot/drivers/spi imports
	periph.io/x/periph/conn/spi: cannot find module providing package periph.io/x/periph/conn/spi
gobot.io/x/gobot/drivers/spi imports
	periph.io/x/periph/host/sysfs: cannot find module providing package periph.io/x/periph/host/sysfs
```

gen2thomas, 2023-07-06:
Hi @jney , please use "gobot/v2/" or after creating your application use "go mod tidy" instead.

gen2thomas, 2023-07-07:
A new documentation is released now, so I will close this issue. Feel free to reopen if we should do something more.

closed 2023-07-07 by gen2thomas

tlindsay, 2023-12-08:
@gen2thomas The [GoBot Getting Started](https://gobot.io/documentation/getting-started/) guide still lists this as the first step:
```
$ go get -d -u gobot.io/x/gobot/...
```

gen2thomas, 2023-12-08:
@tlindsay yes, this is outdated in many places, but there is no time to [fix it](https://github.com/hybridgroup/gobot-site/issues/61).