hybridgroup/gobot #969: Windows Xbox One controller events not detected

Speshl, 2023-07-06:
I am running the scanner.go utility (https://github.com/hybridgroup/gobot/blob/release/platforms/joystick/bin/scanner.go) and I am not able to see button press events from my Xbox one controller.

When I launch the scanner utility with my Xbox controller connected, I see that it shows as a connected controller in the utility but It does not detect button presses.  I see the same issue with my Xbox elite controller.  It does work however with my Logitech g27 steering wheel, and I am able to see events properly.

Any ideas why only the xbox controller would not work?

deadprogram, 2023-07-06:
@Speshl are you on macOS by any chance? It appears the latest SDL versions mappings are different on macOS.

Speshl, 2023-07-06:
No I am on Windows 11.  Also I don't think its the mappings, because the scanner utility is what you use to view all events to figure out the mapping in the first place. The Logitech G27 has no mappings but still shows its events in the scanner.

Speshl, 2023-07-07:
Xbox one and Elite controllers are detected but see no events on Windows 11.  Tried with bluetooth and wired connections.  Works with Logitech G27, Saitek x52, and Thrustmaster stick.  

Interestingly I tried another go joystick repo (https://github.com/0xcafed00d/joystick) and I am seeing the same sets of working and non working controllers. So I don't think it is specific to gobot or SDL2.