hybridgroup/gobot #1007: Support for Sphero BOLT

dennisgove, 2023-10-20:
Sphero sells a device called [Sphero BOLT](https://sphero.com/products/sphero-bolt) which is similar (in appearance) to the SPRK+. However it appears that the Ollie driver code doesn't work with it. In particular, the device characteristics don't match what's defined in [ollie_driver.go](https://github.com/hybridgroup/gobot/blob/release/platforms/sphero/ollie/ollie_driver.go#L30-L40).

These are the characteristics read from the BLE adapter during connection, but all of them appear to have this `unreadable: unknown or unsupported kind: "invalid"` value.

[image]

I'd like to try putting together driver code for the BOLT but I'm not quite sure where to go from here. I haven't been able to find if the BOLT is based on the same Ollie core that the SPRK+ and bb8 appear to be. If not I imagine I'd be starting from scratch but if it is I'm not sure how I'd go about determining which characteristic ID maps to which action (like `wake`, `command`, and `response`).

Do you have advice on where I should go from here? Thank you!

gen2thomas, 2023-10-20:
Hi @dennisgove , I have no experience with sphero toys at all. But there is at least one difference in the python code: The [ToyType](https://github.com/artificial-intelligence-class/spherov2.py/tree/main/spherov2/toy) ends with ".06" for all old devices and with ".075" for the new one. This seems to be the command safe interval. Maybe this is only a timing problem?

My suggestion is to start with a copy of ollie-driver and fix all calls. After that it is possible to optimize the architecture, for the duplicated code parts. Maybe only some constants needs to be different.

If you provide an PR I can support you for the architecture optimization part.

Good luck, Thomas

dennisgove, 2023-10-20:
Thank you, @gen2thomas. I'll peruse that python code and see how far I can get with it. This is a great starting point.

deadprogram, 2023-10-22:
Please see https://github.com/nicholasjackson/sphero-mini for how this can be done using Golang.

Perhaps a new Gobot driver that wraps this package? Or else learn for how it does things and add to the current one? Someone can perhaps take a deep look to decide which of those options seems better.