hybridgroup/gobot #1050: HX711 Support

Juan1ll0, 2023-12-27:
Hi guys,
  Is it possible to write a new driver for HX711 weight cells in Gobot?. Can i use some other device driver supported by Gobot as example?

Thanks.

gen2thomas, 2023-12-27:
Hi @Juan1ll0 , it seems it has a plain serial interface, so it is currently not supported and there is no example driver. As far as I remember there was already an issue to implement a serial interface.

Maybe you can choose [another one](https://www.sparkfun.com/products/15242) or try to use a converter, e.g. to i2c.

Juan1ll0, 2023-12-28:
Thank you very much @gen2thomas I didn't know this kind of sensors. I bought one to test it. I will try to write a driver for Gobot.

Regards.

gen2thomas, 2024-11-01:
Hi @Juan1ll0 , maybe the new serial driver is a start point for your implementation now. Do you have any progress?