hybridgroup/gobot #1077: Write pwm export() failed for id xx "with open gpiochip0"

idalmasso, 2024-03-12:
Trying to write pwm to a LedDriver with a raspberry pi 4 (fresh install of raspbian) fives the error export() failed for id 26 "with open gpiochip0" (trying to use the pin "37" that is actually GPIO 26). 
This error is actually new in v2.3.0 of gobot (I just reverted to 2.2 and everything works fine).

gen2thomas, 2024-03-14:
Hi @idalmasso , thanks for finding and report this issue. I will have a look in the next few days.

gen2thomas, 2024-03-17:
Hi @idalmasso , the PWM usage has changed a bit since 2.3.0.
Until 2.2.0 the pi-blaster was used by default. Now the internal PWMs are used by default and the usage of pi-blaster must be forced by "adaptors.WithPWMUsePiBlaster()". I missed to add this to the migration guide, but it is described in the [README](https://github.com/hybridgroup/gobot/blob/v2.3.0/platforms/raspi/README.md#using-pi-blaster).

If it will not work , please can you provide your program or at least the part for creating the raspi-adaptor and the lines which creates the error output.

Thanks Thomas

idalmasso, 2024-03-18:
Hello @gen2thomas , thanks for your answer. Yes, I just retried with the WithPWMUsePiBlaster and it seems to be working fine, sorry I had to improve my RTFM capabilities, sorry to have made you lose time!

closed 2024-03-18 by idalmasso

gen2thomas, 2024-03-18:
You are welcome, good to hear that it's working now.