hybridgroup/gobot #1087: write /dev/i2c-1: connection timed out

TuringZhu, 2024-08-31:
Hello, I a beginner gobot & raspberry pi.

Now I use [examples/raspi_grove_pi_ultrasonic.go](https://github.com/hybridgroup/gobot/blob/release/examples/raspi_grove_pi_ultrasonic.go) to check with Raspberry Pi Model B and ultrasonic model with HC-SR04.

I have some problems can not be solved.

 HC-SR04 has 4 Pin with VCC,GND,Trig_RX-SCL_I/O, ECHO_DX_SDA but the code just defined `ultrasonicPin = "4"`,  so how can I connect    
my raspberry pi with HC-SR04

and no matter what pin i connect  the code always output `write /dev/i2c-1: connection timed out`. I want to know why.

in addition, I hope to have a website or other chat groups can get other developer's help quickly. I would be extremely grateful.

gen2thomas, 2024-09-16:
Hello @TuringZhu , 

do you have really an I2C ultrasonic Model with the name HC-SR04? Normally this is an analog one. Please try to use this example instead:
https://github.com/hybridgroup/gobot/blob/release/examples/raspi_hcsr04.go

Good luck, Thomas

gen2thomas, 2024-11-01:
Hi @TuringZhu do you have any progress or further questions on that? 

Maybe we can close this issue?

Thanks Thomas

closed 2025-01-11 by gen2thomas