hybridgroup/gobot #1102: sprk+ panic read not permitted

aolney, 2024-11-14:
I'm on

- go version go1.23.3 linux/amd64
- Description:	Ubuntu 22.04.5 LTS

with

```bash
go build bleclient_sprkplus_collision.go 
sudo ./bleclient_sprkplus_collision SK-955D
```

I get this output

```bash
2024/11/14 16:49:02 Initializing connections...
2024/11/14 16:49:02 Initializing connection BLEClient-2F161359000C8B7F ...
2024/11/14 16:49:02 Initializing devices...
2024/11/14 16:49:02 Initializing device SPRKPlus-6FF4B646F0CF43D4 ...
2024/11/14 16:49:02 Robot sprkplus initialized.
2024/11/14 16:49:02 Starting Robot sprkplus ...
2024/11/14 16:49:02 Starting connections...
2024/11/14 16:49:02 Starting connection BLEClient-2F161359000C8B7F...
2024/11/14 16:49:05 Starting devices...
2024/11/14 16:49:05 Starting device SPRKPlus-6FF4B646F0CF43D4...
2024/11/14 16:49:05 Starting work...
panic: Read not permitted

goroutine 280 [running]:
gobot.io/x/gobot/v2/drivers/ble/sphero.(*OllieDriver).initialize.func2()
	/home/aolney/go/pkg/mod/gobot.io/x/gobot/v2@v2.4.0/drivers/ble/sphero/sphero_ollie_driver.go:237 +0x65
created by gobot.io/x/gobot/v2/drivers/ble/sphero.(*OllieDriver).initialize in goroutine 1
	/home/aolney/go/pkg/mod/gobot.io/x/gobot/v2@v2.4.0/drivers/ble/sphero/sphero_ollie_driver.go:234 +0x14a
```

I can avoid this error by commenting out references to `ball`, e.g.
``` go
ball := sphero.NewSPRKPlusDriver(bleAdaptor)
```
In other words, if I have a robot with just the bleclient, there is no error. So the problem seems to be with NewSPRKPlusDriver.

Any suggestions appreciated :) This is my first `go` project

gen2thomas, 2024-11-15:
Hi @aolney , thanks for adding this issue. Unfortunately I had no hardware for testing this after I have done some rework. Only the existing tests helped me a little bit, but you can see there are not that much.

Back to your finding:
The panic comes from a go-routine in line 234 of the base driver sphero_ollie_driver.go (asynchronous execution each 100ms). Surprisingly it calls ReadCharacteristic(), but drops the return value except the error.

It seems you are using the [example](https://github.com/hybridgroup/gobot/blob/release/examples/bleclient_sprkplus_collision.go). So the initialize() function is called from your main in line [47](https://github.com/hybridgroup/gobot/blob/4a19dee361071c18db7a9b9cb724dc52de86ae12/examples/bleclient_sprkplus_collision.go#L47) (robot.Start()).

My first idea is to drop this go function completely, because in line [220](https://github.com/hybridgroup/gobot/blob/4a19dee361071c18db7a9b9cb724dc52de86ae12/drivers/ble/sphero/sphero_ollie_driver.go#L220) we subscribe on changes of "responseChara". If a cyclic call is needed, maybe the device is not ready until this call.

For this a second idea comes up, change the [go function](https://github.com/hybridgroup/gobot/blob/4a19dee361071c18db7a9b9cb724dc52de86ae12/drivers/ble/sphero/sphero_ollie_driver.go#L234C1-L241C5), so the error is logged only and increase sleep: 	
```
go func() {
		for {
			if _, err := d.Adaptor().ReadCharacteristic(responseChara); err != nil {
				// panic(err)
                                fmt.Printf("error issue 1102: %v",err)
			}
			// time.Sleep(100 * time.Millisecond)
                        time.Sleep(1000 * time.Millisecond)
		}
	}()
```

I am looking forward to your feedback, Thomas

aolney, 2024-11-26:
Thanks for those suggestions @gen2thomas 

Deleting the function seems to be the way to go. If you modify it with the message as suggested, it just seems to print out the message forever. 

After deleting, I got correct behavior with `go run examples/bleclient_bb8-collision.go SK-955D`

gen2thomas, 2024-11-27:
Hi @aolney , thanks for your detailed answer. I will re-open this issue and mark it as a bug. The fix can be the removing of the complete go function. If you encounter a related problem when you make further tests, please add information to this issue.

gen2thomas, 2025-02-16:
fixed in v2.5.0

closed 2025-02-16 by gen2thomas